FPGA-MAFIA/fpga_mafia #5: Make sure you can simulate Systemverilog

amichai-bd, 2022-08-07:


amichai-bd, 2022-09-05:
- [ ] @danielk532
 - [ ]  @yeonatanPerelman
 - [ ]  @ShmuelSfez

danielk532, 2022-09-06:
I run a simulation for NAND gate with and without gui - 
[image]
[image]

amichai-bd, 2022-11-06:
@yeonatanPerelman @ShmuelSfez 
Reminder

yeonatanPerelman, 2022-11-21:
we fixed all our problems

closed 2022-11-21 by yeonatanPerelman

amichai-bd, 2022-11-21:
@yeonatanPerelman - Next time please document what was the issue you faced to complete the task.
For example  in this case:
- Make sure first to choose what signals you want to view.   
A simple way is to select hierarchy on the left side, then press the \<ctrl\>+'w' to "bring" all the hierarchy signals. 
- Then press the "Restart" button & Press the Run Button
[image]

- Alternatively, you may use the "VSIM" terminal to run all the commands:
[image]

noamsabb, 2022-11-24:
i ran a simulation for XOR gate, i get good results without gui but can't see any waves in gui, look like the code is not running
[image]
[image]

amichai-bd, 2022-11-24:
Did you try and press this:
[image]