FPGA-MAFIA/fpga_mafia #11: find the root cause and solve the slt/sltu bug in the CPU

amichai-bd, 2022-10-03:
Please mark yourself as done once you pushed a solution to a branch
Once you are all done, you may close the issue as completed

- [ ] @danielk532 
- [ ] @yeonatanPerelman 
- [ ] @ShmuelSfez

amichai-bd, 2022-11-06:
Did any of you find why the slt/sltu instructions did not work as expected?

danielk532, 2022-11-06:
[image]

After a little debug, the output seems correct:)

amichai-bd, 2022-11-06:
You fixed the bug?
Or just realized that the code was correct the whole time?

danielk532, 2022-11-06:
I fixed it, but I didnt pushed my solution yet. I need some help with that.

amichai-bd, 2022-11-06:
Glad you were able to find the bug!
Will ping you on WhatsApp later

yeonatanPerelman, 2022-11-21:
fixed by daniel and alredy merged to main branch
[image]

closed 2022-11-21 by yeonatanPerelman