FPGA-MAFIA/fpga_mafia #14: Add a mesh directory under the source directory

amichai-bd, 2022-11-06:
@yeonatanPerelman & @ShmuelSfez 
Add a directory called "mesh" under the "source" directory.
In the directory add a file called "README.md"
In the file, please write a short description of the project you are starting as you understand it. (1-3 paragraphs)

amichai-bd, 2022-11-15:
+ mini_core directory under source

The same mini_cor & mesh under verif. 
verif
| mini_core
|----|-- tb
|----|-- tests

Add README.

amichai-bd, 2022-11-20:
@yeonatanPerelman @ShmuelSfez 
This task should be simple - no need for simulation or anything.
Let me know what holds you back from this task.

yeonatanPerelman, 2022-11-21:
I created the mini_core directory with Amichai help.
I will do with you the mesh directory @ShmuelSfez

ShmuelSfez, 2022-11-21:
added mesh dirs and README

closed 2022-11-21 by ShmuelSfez