FPGA-MAFIA/fpga_mafia #22: create a alive tests using for your core.

amichai-bd, 2022-11-15:
- [ ] @danielk532 
- [ ] @yeonatanPerelman  & @ShmuelSfez 

You may use the current implementation we designed together.

I would like to be able to read your README for each model and execute your "local" core.
Eventually, you will modify the core to be the one we need. (3/7 stage Pipeline)

amichai-bd, 2022-11-24:
This means you need to copy the design we already have into your directory under source, then add your tb under verif.

Let me know if you have any questions on this.

amichai-bd, 2022-11-27:
@yeonatanPerelman @ShmuelSfez - Use the RVC_ASAP as reference (not the Single-cycle we did together)
Add the tracker there to complete the other issue asigned to you

amichai-bd, 2022-11-30:
@danielk532 - I see you marked yourself as done - but I don't see you're content in the main or your pull request.

ShmuelSfez, 2022-12-03:
the test is alive!!!

closed 2022-12-03 by ShmuelSfez