FPGA-MAFIA/fpga_mafia #47: Cache - Add a simple TB & DUT to the Cache directory

amichai-bd, 2022-12-01:
Design a simple "xor" module + tracker
source/cache
verif/cache/tb
modelsim/cache
target/cache

noamsabb, 2022-12-05:
Hi @amichai-bd  I'm not sure I know what are DUT and tracker
I added the module, list and TB files to the appropriates cache directories and succeed to compile and run simulation 
[image]
[image]

amichai-bd, 2022-12-06:
DUT - Device under test. the Component that is being created and tested by the verification.
The tracker - Is a generated log file that tracks some elements in your design.

Please see the big_core_tb as a reference for trackers. 

Once you are ready to "commit" please add to your commit message "close #47" which will close this issue.

amichai-bd, 2022-12-06:
If this done, please close this issue.

amichai-bd, 2022-12-17:
Done

closed 2022-12-17 by amichai-bd