FPGA-MAFIA/fpga_mafia #49: mini_core - "memory wrap"

amichai-bd, 2022-12-04:
step 1:
Create the i_mem + d_mem as you see here:
https://github.com/amichai-bd/riscv-multi-core-lotr/tree/master/source/gpc_4t/rtl

Add a new module called "mem_wrap" which holds 2 modules:
- inst_mem          
  - fabric : read/write               
  - Core : read
- data_mem 
  - fabric : read/write               
  - Core : read/write

Please make the standard interface to match the FPGA style - see reference here:
https://github.com/amichai-bd/riscv-multi-core-lotr/blob/master/source/gpc_4t/rtl/d_mem_wrap.sv

Please note: the FPGA std memory is only 32-bit aligned!
You must ensure any non-32-bit aligned request will read/write correctly!
This is done by shifting the data & enable it correctly!

amichai-bd, 2022-12-20:
Make sure you finish the trackers correctly./
When done please close the issues using the commit

amichai-bd, 2022-12-29:
Was this done?
Please make sure to close this issue using the commit "close \#\<number\>"

amichai-bd, 2023-01-22:
@ShmuelSfez - make sure to pull origin main, fix any merge conflicts and push your latest!

ShmuelSfez, 2023-02-27:
done.

closed 2023-02-27 by ShmuelSfez