FPGA-MAFIA/fpga_mafia #104: big_core - find how to pass a variable between the linker and package

danielk532, 2023-01-08:
find how to pass a variable between the linker and package

amichai-bd, 2023-01-10:
I was able to pass a variable to the linker from the GCC - this will allow us to define a parameter that will be used both for the GCC SW linker & we can use it to override the parameter in the systemverilog package using the "pvalue" option that can overrides parameters.

We will need to make sure we do the same once we get to FPGA.

This picture shows the "-Wl,--defsym=" that passes variables. - 
[image]
you will need to update the build to support a new flag that will override both HW mem offsets & sizes.

Dont forget to pull from origin before starting to get my chages

amichai-bd, 2023-01-24:
Closing this issue.
In issue #107 you will add the global defines per core configuration.
There you will use the defines to override the mem length & offset

closed 2023-01-24 by amichai-bd