FPGA-MAFIA/fpga_mafia #121: cache - design the merge buffer to allow multiple writes to same cl

amichai-bd, 2023-01-19:
Make sure not to allocate a new entry when core2cache request matches the existing cl address in tq.
- Use the merge buffer to write the new data.
- let the pipe know not to send fm fill. (due to the hit in the merge buffer)
- in case of read after write.
- set the "rd" bit on the merge buffer so that when the fill eventually enters the pipe, it will continue to the cache2core response.

amichai-bd, 2023-03-02:
Make sure we have a test for this,
Please comment here what is test name for this scenario.
if we don't have it - create it and make sure it behaves as expected.

noamsabb, 2023-03-03:
The test we wrote for this scenario 

[image]

We can see from the pipe_trk on this test that the fill cancel is not working/enable and we get a fill for the others write and read request even if we are on the same CL

[image]

Also it look like the merge buffer is not getting update like it would after every write miss and it look like its erasing the old values of the write misses instead of keeping them in the mb

noamsabb, 2023-03-03:
We would have wanted only one fill to happen, to bring the actual CL from FM and for every write_req to write their data to the mb 

[image]

instead of that we can see that the second fill in line 19 (that is not supposed to happen) erased the data wrote in the mb when the first wr miss happen

Also the second fill is for address 1010c so it is for the last wr_req not even the second one

I noticed we didn't added signal for wr_req_hit_mb like we did for rd_req so i tried to add them and adding the same features that we added for rd_req_hit_mb but it still not resolving the problem

amichai-bd, 2023-03-04:
Please push your changes to your branch so I can review it..

closed 2023-03-09 by noamsabb