FPGA-MAFIA/fpga_mafia #133: big_core - enable the VGA memory space & create test

amichai-bd, 2023-01-24:
1. Instantiate the VGA unit - https://github.com/amichai-bd/riscv-multi-core-lotr/blob/master/source/lotr/rtl/vga_ctrl.sv
2. Add an EOT (end-of-test) sequence task to print the content of the VGA memory to a file
[image]
3. create a display test & include the VGA library. 
    https://github.com/amichai-bd/riscv-multi-core-lotr/blob/master/apps/scripts/gpc_compile_link/graphic.h
    - Print hello-world.
    - use the ```\n``` to write to multiple lines
    - move the user to a specific location and print there
4. instead of creating the VGA CRs (for the curser), allocate a memory address with a VGA library.

amichai-bd, 2023-02-12:
- Note: the graphic lib is enabled 
- See reference in SC core - but create the VGA mem space like this reference:
Instantiate the VGA unit - https://github.com/amichai-bd/riscv-multi-core-lotr/blob/master/source/lotr/rtl/vga_ctrl.sv

danielk532, 2023-04-30:
Yes, it looks like it's done. 
You did it with me

closed 2023-04-30 by danielk532