FPGA-MAFIA/fpga_mafia #269: fifo_arb-  valid_arb to the arbiter isn't good

yeonatanPerelman, 2023-05-11:
the valid arb should help the RR to choose the next winner to pop. not a general valid_arb.

[image]

amichai-bd, 2023-05-12:
I will note the with this current implementation, we found in the router test a case that a transaction is "lost".. we had a typo in the TB where the "ready" was stock at 0. And the fifo arb was pushed & popped, but the transaction never came out due to the wrong use of valid_arb.

The way the arb need to work is a round Robin with a "mask" on the candidates.

1. Please create a test that holds the ready low, and only after couple of cycles releases it back to high - this test will fail due to he bug described.
2. Once we have the failing test, we will work on the RTL fix - re writing the RR with the correct mask.

amichai-bd, 2023-05-13:
The RTL was already fixed in the branch "abd_fix_fifo_arb_ready_logic"
Please create the test - make sure we fail, then merge the abd_fix_fifo_arb_ready_logic branch and see if it passes

closed 2023-05-14 by amichai-bd