FPGA-MAFIA/fpga_mafia #280: add more complicated sequences to fabric. support the env and checkers for it.

ShmuelSfez, 2023-05-21:
1. add full fabric seq - create a stress seq, all tiles full.
2. add support for tb - not generate a transaction in to a tile that already has a transactions in it.
3. add DI checker- create a generic DI checker for all cases, check that the transaction did get from source to target for each trans.
4. create more generic TB (maybe instead of the `ASSIGN macro).

amichai-bd, 2023-05-21:
@ShmuelSfez regarding "2" - to be more precise, "not generate a new transaction from a tile that has back pressure from the router"

ShmuelSfez, 2023-05-21:
@amichai-bd - can i assume for my checker that if a transaction is get into a local tile then it is the final target from perspective of the source that send the request?

amichai-bd, 2023-05-21:
Yes

ShmuelSfez, 2023-07-05:
done

closed 2023-07-05 by ShmuelSfez

amichai-bd, 2023-07-06:
@ShmuelSfez - looking at the fabric level0, I dont see param overrides. is that expected?