FPGA-MAFIA/fpga_mafia #294: FABRIC - create a simple Rd after Wr from 1_1 to 3_3 - make sure data is correct

amichai-bd, 2023-06-04:
Please create a simple test that can write from 1_1 to a specific address in 3_3.
Then it sends a read request to the same address. (1_1->3_3)

We expect 3 transactions in total:
1) wr (1_1 -> 3_3)
2) rd (1_1 -> 3_3)
3) rd_rsp (3_3->1_1)

add a checker that the data is correct

ShmuelSfez, 2023-06-06:
i created a direct test for it, currentley im not getting the data by RD, need to check why.

amichai-bd, 2023-06-06:
Great! might be an issue in the mem_wrap.
Also - the mem_wrap supports specific memory offset & size.
You must make sure your random mem address offset is In the valid range

@ShmuelSfez  Please add it to the level_fail so i can give it a look.

ShmuelSfez, 2023-07-05:
done

amichai-bd, 2023-07-06:
@ShmuelSfez  - Looking at the fabric level0, i don't see any param override.. Is this expected?

ShmuelSfez, 2023-07-07:
yes. this is a direct test with no randomization at all.

amichai-bd, 2023-07-08:
O ya.. i meant to write this on the other BP tests...
Is the BP tests part of the level0?