FPGA-MAFIA/fpga_mafia #356: possible additions in gui_build.py

roman012285, 2023-09-29:
I hope I'm not overstepping my boundaries by making changes to files that may not be within my scope, but I thought you might want to consider adding the '-gui' option to the 'gui_build' file. In any case, I experimented with it a bit and included the option. I also updated the '-app' description to mention that you can run it from assembly code, not just C. I saved these changes **on my computer only** without modifying my local repository so it won't effect the global in the future.

[image]

amichai-bd, 2023-09-30:
1) this is an open source project - you are welcome to change, add, modify whatever you find fit. 

It will go through the "code review" process so I won't allow anything I don't approve into the main branch.

2) the changes you mentioned sound great I will love to add them to the main so please do add a PR so I can review and approve so we all enjoy you contribution!

amichai-bd, 2023-09-30:
I will mention that there are many other fixes i would like to add into the build itself and not only the "build_gui".

Having you working on them would be amazing!

roman012285, 2023-09-30:
Sure, I will be happy to.

closed 2023-09-30 by roman012285