FPGA-MAFIA/fpga_mafia #358: The "new" big_core - by Roman

amichai-bd, 2023-10-01:
# Create a 6-stage pipe-line CPU.
- 100: fetch
- 101: decode
- 102: execute
- 103: mem_access
- 104: mem_access_part2
- 105: write_back  

**as you can see, we have 2-cycle latency on the mem access!!! this is due to the Cachre lookup latency we want to incorporate later.**
**Note:  the I_MEM is still 1 cycle latency!!**
**Document the units in the wiki!!!!**

# Use the coding style of the mini_core ase base line for everything(!!)
- RTL - struct, enum, macros, ctrl, etc..
- verification - reference model (RF tracker) 

# Add a second checker for the Data Integrity (expected vs actually memory read + write)
- Similar to the RF checker, but will track both the rd/wr (loads/store) (and not only the write like we did in the RF DI)

# Once we have the 6-stage CPU working:
### Support CSR
- use reference from the big_core to support CSR
- Support SW exception/interrupt 
- Support counter for PMON (performance monitor)

### Support VGA
- We might want to redesign the SW & the HW of the VGA controller, the current implementation is a little crazy and we want to simplify it.
- Supprt VGA RGB mode!

### Support the UART 
- important to be able to load programs to the core from "host"
- currently complemented in the big_core

### support keyboard
- this is WIP by big_core should be done my next week

### Write a simple "OS-like" to wrap the program and handle the exceptions
- 

### integrate Cache 
- make sure we are able to "Back pressure"  so we can have non-deterministic memory latency access. (this already exists in mini_core)
- this will allow both fabric access + cache access which are both non-deterministic.
- Replace the D_MEM (no the I_MEM) with the d_cache unit. - make sure to add support for "Far memory" in simulation.

amichai-bd, 2023-12-25:
Seems a lot of this is done!
Let's try and break it down: 

- [ ] Create a 6-stage pipe-line CPU. (Use the coding style of the mini_core ase base line for everything(!!))
- [ ] Add a second checker for the Data Integrity (expected vs actually memory read + write)
- [ ] Support CSR (+ exception)
- [ ] Support VGA
- [ ] Support the UART
- [ ] support keyboard
- [ ] Write a simple "OS-like" to wrap the program and handle the exceptions
- [ ] integrate Cache

amichai-bd, 2024-01-15:
@roman012285 
Lets break this issue down into multiple smaller issues
Please close this issue and create hte missing issues.
May copy the content of this issue to the new issues

roman012285, 2024-01-15:
the left parts of the issue been devided into small issue: #207, #430, #433, #436

closed 2024-01-15 by roman012285