FPGA-MAFIA/fpga_mafia #364: fail to execute ` ./build.py -cfg mini_rv32i -dut mini_core -test alive -app -hw -sim`

roman012285, 2023-10-05:
Data integrity test fail!

Notes:
1. This test **also fails** with older version `build.py` (before adding -cfg flag) and crt0.s that looks lilke `crt0_mini_rv32`
2. This test **pass** when **only** adding `.org100` at the end of  `crt0_mini_rv32` file

amichai-bd, 2023-10-05:
I will get to give this a look only next week.

To debug this:
1) create a super simple test. (alive)
2) Compare the execution of the reference vs RTL
3) see which one is correct

roman012285, 2023-10-06:
- milestone 
I am running the following code
```
int main(){
   int x =5;
    return 0;
}
```

I am comparing between two `crt0_mini_rv32i':

### option 1 (the one that causes the integrity fail)
```
_start:
  .global _start
  .org 0x00
  nop                       
  nop                       
  nop                       
  nop                       
  nop                       
reset_handler:
  mv  x1, x0
  mv  x2, x1
  mv  x3, x1
  mv  x4, x1
  mv  x5, x1
  mv  x6, x1
  mv  x7, x1
  mv  x8, x1
  mv  x9, x1
  mv x10, x1
  mv x11, x1
  mv x12, x1
  mv x13, x1
  mv x14, x1
  mv x15, x1
  mv x16, x1
  mv x17, x1
  mv x18, x1
  mv x19, x1
  mv x20, x1
  mv x21, x1
  mv x22, x1
  mv x23, x1
  mv x24, x1
  mv x25, x1
  mv x26, x1
  mv x27, x1
  mv x28, x1
  mv x29, x1
  mv x30, x1
  mv x31, x1
  /* stack initialization */
  la   x2, _stack_start

  jal x1, main  //jump to main
  ebreak        //end
  .section .text  

```

` at that case iam looking at `trk_write_reg_ref.log'. I am adding the relevant part that cause the difference:

[image]

at time 540ns/550ns it does two same operations on PC = 0x0a0. That address represents the <main>
I mean that in time 540ns/550ns PC points on main but the main starts to be executed only on 550ns

### Option 2 (the working 1)
```
_start:
  .global _start
  .org 0x00
  nop                       
  nop                       
  nop                       
  nop                       
  nop                       
reset_handler:
  mv  x1, x0
  mv  x2, x1
  mv  x3, x1
  mv  x4, x1
  mv  x5, x1
  mv  x6, x1
  mv  x7, x1
  mv  x8, x1
  mv  x9, x1
  mv x10, x1
  mv x11, x1
  mv x12, x1
  mv x13, x1
  mv x14, x1
  mv x15, x1
  mv x16, x1
  mv x17, x1
  mv x18, x1
  mv x19, x1
  mv x20, x1
  mv x21, x1
  mv x22, x1
  mv x23, x1
  mv x24, x1
  mv x25, x1
  mv x26, x1
  mv x27, x1
  mv x28, x1
  mv x29, x1
  mv x30, x1
  mv x31, x1
  /* stack initialization */
  la   x2, _stack_start

  jal x1, main  //jump to main
  ebreak        //end
  .section .text  
  nop
```

lets look again at `trk_write_reg_ref.log'. I am adding the relevant part that cause the difference:

[image]

Now the <main> starts at PC = 0xa4 (because we added the NOP) and in that case we dont point twice on main and starts to execute main at the moment PC points on it. See 540ns and 550ns

amichai-bd, 2023-10-06:
Again - i think you are getting the same conclusion i had ->mini_core is the issue.

But it will be easy to prove by looking at the _elf.txt and the actual execution.
I will give it a quick look now see what i can find

amichai-bd, 2023-10-06:
I changed the DI integrity a little to print all the reference writes instead of the RTL writes for the compare loop:
[image]
 
Seems like there is an issue with the reference model.

This is the code that needs to be executed:
```
...
  4c:	00008793          	mv	a5,ra
  50:	0001f117          	auipc	sp,0x1f
  54:	fb010113          	addi	sp,sp,-80 # 1f000 <_stack_start>
  58:	008000ef          	jal	ra,60 <main>
  5c:	00100073          	ebreak

00000060 <main>:
  60:	ff810113          	addi	sp,sp,-8
  64:	00812223          	sw	s0,4(sp)
  68:	00810413          	addi	s0,sp,8
  6c:	00500793          	li	a5,5
  70:	fef42c23          	sw	a5,-8(s0)
  74:	00000793          	li	a5,0
  78:	00078513          	mv	a0,a5
  7c:	00412403          	lw	s0,4(sp)
  80:	00810113          	addi	sp,sp,8
  84:	00008067          	ret
```
See the checker: 
```
# ===============================
# Starting data integrity test
# ===============================
# ref_rf_write_history size = 30
# rf_write_history size     = 25
#  >> rf_write_history[0] Match: time: 190, PC: 00000014, RegDsd:  1, Data: 00000000
#  >> rf_write_history[1] Match: time: 200, PC: 00000018, RegDsd:  2, Data: 00000000
#  >> rf_write_history[2] Match: time: 210, PC: 0000001c, RegDsd:  3, Data: 00000000
#  >> rf_write_history[3] Match: time: 220, PC: 00000020, RegDsd:  4, Data: 00000000
#  >> rf_write_history[4] Match: time: 230, PC: 00000024, RegDsd:  5, Data: 00000000
#  >> rf_write_history[5] Match: time: 240, PC: 00000028, RegDsd:  6, Data: 00000000
#  >> rf_write_history[6] Match: time: 250, PC: 0000002c, RegDsd:  7, Data: 00000000
#  >> rf_write_history[7] Match: time: 260, PC: 00000030, RegDsd:  8, Data: 00000000
#  >> rf_write_history[8] Match: time: 270, PC: 00000034, RegDsd:  9, Data: 00000000
#  >> rf_write_history[9] Match: time: 280, PC: 00000038, RegDsd: 10, Data: 00000000
#  >> rf_write_history[10] Match: time: 290, PC: 0000003c, RegDsd: 11, Data: 00000000
#  >> rf_write_history[11] Match: time: 300, PC: 00000040, RegDsd: 12, Data: 00000000
#  >> rf_write_history[12] Match: time: 310, PC: 00000044, RegDsd: 13, Data: 00000000
#  >> rf_write_history[13] Match: time: 320, PC: 00000048, RegDsd: 14, Data: 00000000
#  >> rf_write_history[14] Match: time: 330, PC: 0000004c, RegDsd: 15, Data: 00000000
#  >> rf_write_history[15] Match: time: 340, PC: 00000050, RegDsd:  2, Data: 0001f050
#  >> rf_write_history[16] Match: time: 350, PC: 00000054, RegDsd:  2, Data: 0001f000
#  >> rf_write_history[17] Match: time: 360, PC: 00000058, RegDsd:  1, Data: 0000005c
#  >> rf_write_history[18] Match: time: 390, PC: 00000060, RegDsd:  2, Data: 0001eff8
#  >> rf_write_history[19] Match: time: 410, PC: 00000068, RegDsd:  8, Data: 0001f000
#  >> rf_write_history[20] Match: time: 420, PC: 0000006c, RegDsd: 15, Data: 00000005
#  >> rf_write_history[21] Match: time: 440, PC: 00000074, RegDsd: 15, Data: 00000000
#  >> rf_write_history[22] Match: time: 450, PC: 00000078, RegDsd: 10, Data: 00000000
#  >> rf_write_history[23] Match: time: 460, PC: 0000007c, RegDsd:  8, Data: 00000000
#  >> rf_write_history[24] Match: time: 480, PC: 00000080, RegDsd:  2, Data: 0001f000
#  >> rf_write_history[25] Mismatch!!
# ** Error: ERROR: rf_write_history mismatch
#    Time: 490 ps  Scope: mini_core_tb.di_register_write File: ../../../verif/mini_core/tb/mini_core_tasks.vh Line: 80
#       ref_rf_write_history[25] =   {time: 440, Pc: 00000060, RegDst:  2, Data: 0001eff8}
#       rf_write_history    [25] =   {time: x, Pc: xxxxxxxx, RegDst:  x, Data: xxxxxxxx}
#  >> rf_write_history[26] Mismatch!!
# ** Error: ERROR: rf_write_history mismatch
#    Time: 490 ps  Scope: mini_core_tb.di_register_write File: ../../../verif/mini_core/tb/mini_core_tasks.vh Line: 80
#       ref_rf_write_history[26] =   {time: 450, Pc: 00000060, RegDst:  2, Data: 0001eff0}
#       rf_write_history    [26] =   {time: x, Pc: xxxxxxxx, RegDst:  x, Data: xxxxxxxx}
#  >> rf_write_history[27] Mismatch!!
# ** Error: ERROR: rf_write_history mismatch
#    Time: 490 ps  Scope: mini_core_tb.di_register_write File: ../../../verif/mini_core/tb/mini_core_tasks.vh Line: 80
#       ref_rf_write_history[27] =   {time: 460, Pc: 00000060, RegDst:  2, Data: 0001efe8}
#       rf_write_history    [27] =   {time: x, Pc: xxxxxxxx, RegDst:  x, Data: xxxxxxxx}
#  >> rf_write_history[28] Mismatch!!
# ** Error: ERROR: rf_write_history mismatch
#    Time: 490 ps  Scope: mini_core_tb.di_register_write File: ../../../verif/mini_core/tb/mini_core_tasks.vh Line: 80
#       ref_rf_write_history[28] =   {time: 470, Pc: 00000060, RegDst:  2, Data: 0001efe0}
#       rf_write_history    [28] =   {time: x, Pc: xxxxxxxx, RegDst:  x, Data: xxxxxxxx}
#  >> rf_write_history[29] Mismatch!!
# ** Error: ERROR: rf_write_history mismatch
#    Time: 490 ps  Scope: mini_core_tb.di_register_write File: ../../../verif/mini_core/tb/mini_core_tasks.vh Line: 80
#       ref_rf_write_history[29] =   {time: 480, Pc: 00000060, RegDst:  2, Data: 0001efd8}
#       rf_write_history    [29] =   {time: x, Pc: xxxxxxxx, RegDst:  x, Data: xxxxxxxx}
# ** Error: ERROR: rf_write_history size mismatch
#    Time: 490 ps  Scope: mini_core_tb.di_register_write File: ../../../verif/mini_core/tb/mini_core_tasks.vh Line: 88
# Data Integrity final status: Data integrity test failed - rf_write_history size mismatch
# ===============================
```

As you can see, up to instruction ``` 80:	00810113          	addi	sp,sp,8```
Everything is great. and the expectation at 84 is to jump back to the "ra" register, and then execute the ebreake.
(which means no more RF writes.)

But in the DI checker, the reference model started executing pc=60 multiple times.
Which creates the diff.

# Solution:
1) fast and easy: simply add a nop after the break (of add 2 break)
2) Go over the reference model, see if we can suppress any new writes after the ebreak

roman012285, 2023-10-06:
It seems that the problem has fixed but please verify that I am write cause I changed the code.
In rv32_rwf_sv I comment line 85 and changes it with `assign` statement as seen in line 84.
I tested with `mini_core` and `big core` and it works. I also tested `mini_core alive.c in big_core` to get an error as expected and we also got error with the new line.

[image]

### I also noticed that `./build.py` complements the test name even if its not exists, for example:
-test a works!!??

[image]

closed 2023-10-11 by roman012285