FPGA-MAFIA/fpga_mafia #423: sc_core - screen is broken, and there is no reference model in TB

amichai-bd, 2024-01-14:
2 things to fix:
1) add a reference model and EOT check to make sure IP is running correctly
2) Make sure the display works. (have a dedicated memory region for VGA space)

roman012285, 2024-01-14:
I fixed the screen. But the code dosn't looks that good (in readability) because of the Dmem naming and the fact that memory of the core is located inside the TB. Before I make PR please take a look at the `sc_core_tb.sv` file:
```
//-----------------------------------------------------------------------------
// Title            : single cycle Core TB
// Project          : 
//-----------------------------------------------------------------------------
// File             : sc_core_tb.sv
// Original Author  : Amichai
// Code Owner       : Amichai
// Created          : 2/2023
//-----------------------------------------------------------------------------
// Description :
// simple test bench
// (1) generate the clock & rst. 
// (2) load backdoor the I_MEM & D_MEM.
// (3) End the test when the ebrake command is executed
//-----------------------------------------------------------------------------
`include "macros.sv"

module sc_core_tb ;
import sc_core_pkg::*;

logic        Clk;
logic        Rst;
logic [31:0] Instruction;
logic [31:0] DMemAddress;
logic [31:0] DMemData   ;
logic [3:0]  DMemByteEn ;
logic        DMemWrEn   ;
logic        DMemRdEn   ;
logic [31:0] DMemRspData;
logic  [7:0] IMem       [I_MEM_MSB : 0];
logic  [7:0] NextIMem   [I_MEM_MSB : 0];
logic  [7:0] DMem       [D_MEM_SIZE + D_MEM_OFFSET - 1 : D_MEM_OFFSET];
logic  [7:0] NextDMem   [D_MEM_SIZE + D_MEM_OFFSET - 1 : D_MEM_OFFSET];
logic  [7:0] VgaMem     [VGA_MEM_REGION_ROOF : VGA_MEM_REGION_FLOOR];
logic  [7:0] NextVgaMem [VGA_MEM_REGION_ROOF : VGA_MEM_REGION_FLOOR];
logic [31:0] Pc;

`include "sc_core_tasks.vh"

//=========================================
// Instantiating the sc_core core
//=========================================

 sc_core sc_core (
    .Clk            (Clk),                      
    .Rst            (Rst),                      
    .Pc             (Pc),         // To I_MEM   
    .Instruction    (Instruction),// From I_MEM 
    .DMemData       (DMemData),   // To D_MEM   
    .DMemAddress    (DMemAddress),// To D_MEM   
    .DMemByteEn     (DMemByteEn), // To D_MEM   
    .DMemWrEn       (DMemWrEn),   // To D_MEM   
    .DMemRdEn       (DMemRdEn),   // To D_MEM   
    .DMemRspData    (DMemRspData) // From D_MEM 
);
//======================
//  instruction memory
//======================
assign Instruction[7:0]   = IMem[Pc+0];
assign Instruction[15:8]  = IMem[Pc+1];
assign Instruction[23:16] = IMem[Pc+2];
assign Instruction[31:24] = IMem[Pc+3];
//===========================
//  Data and VGA memory Write
//===========================
logic DmemVgaHit;
assign DmemVgaHit = (DMemAddress >= D_MEM_OFFSET && DMemAddress < D_MEM_SIZE + D_MEM_OFFSET) ? 1'b1 : 1'b0;
always_comb begin
[... 109 lines cut ...]
            for (int k = 0 ; k < 320; k = k+4) begin // Words
                for (int l = 0 ; l < 8; l = l+1) begin // Bits  
                    draw = (VgaMem[VGA_MEM_REGION_FLOOR+k+j+i][l] === 1'b1) ? "x" : " ";
                    $fwrite(fd1,"%s",draw);
                end        
            end 
            $fwrite(fd1,"\n");
        end
    end
endtask

rv32i_ref
# (
    .I_MEM_LSB (I_MEM_OFFSET),
    .I_MEM_MSB (I_MEM_MSB),
    .D_MEM_LSB (D_MEM_OFFSET),
    .D_MEM_MSB (D_MEM_MSB)
)  rv32i_ref (
.clk    (Clk),
.rst    (Rst),
.run    (1'b1) // every time the run is set, the next instruction is executed
);

endmodule //big_core_tb
```

amichai-bd, 2024-01-14:
I think it looks fine Great job!
FYI - you have a bug or a typo in the naming 
[image]
when dmemVGAHit you are accessing the Dmem, but if its "!DmemVGAHit" you are accesing the VGA mem.

roman012285, 2024-01-14:
actually its what I meant. Its working. I mean that when DmemVgaHit = 1 we accsees Dmem.
So I will make PR and close the issue

amichai-bd, 2024-01-14:
This is strange to me..
DmemVgaHit = 1 we accsees Dmem?? 
I think if you "VGAHit" you should access VGA mem.

amichai-bd, 2024-01-14:
Maybe its better to add 2 conditions:
DMemHit = <range>
DMemVgaHIt = <range>

And use each correctly..
This may be simpler to understand.

roman012285, 2024-01-14:
I will change it, no problem. Just because the name starts with Dmem I decided to assign '1' when we want to use that region. But never mind I will change it

closed 2024-01-15 by roman012285