FPGA-MAFIA/fpga_mafia #430: core_rrv - support the PS2 keybord interface

amichai-bd, 2024-01-14:
See reference from the BIG_CORE.
- We should create a more robust verification for the PS2. including a simple foundation for creating "fscan" ("cin") 
- For now, use the RTL of the BIG_CORE_KBD without changing the content (unless you find a bug)
- I suspect it will be tricky to get the same behavior across the DUT and the reference model -> need to come up with a "new type of verification"
- Please break down the task and ask questions here.

roman012285, 2024-01-14:
got it!

roman012285, 2024-02-23:
hi Daniel, nice to meet you. I am working on PS2 the project you started
1. is it possible to ask you some questions?
2. Amihai told me you are writing a documentation on PS2, I would like to use it so I wont bother you with questions!
Thanks, roman

roman012285, 2024-02-28:
That issue was spitted into smaller issues. They belong to  `PS2 keyboard implementation` project.

closed 2024-02-28 by roman012285