FPGA-MAFIA/fpga_mafia #467: Timer interrupt bug - shorter intervals are not triggering as much as needed

amichai-bd, 2024-02-04:
@ roman012285 - we still have some issues with the timer interrupts.  
I suspect the issue is in the mechanism that enables/disables the interrupt handler.

The tests: 
``` python build.py -dut core_rrv -cfg core_rrv_boot_trap -top core_rrv_no_ref_tb -tests "bubblesort_lfsr" -app -hw -sim```

found the issue when when changed the values of the "TIMER_INTERRUPT_INTERVAL". ("interrupt_handler.h")
See example:
For: ``` #define TIMER_INTERRUPT_INTERVAL          0x00000500  ```
Got:
[image]  

For: ``` #define TIMER_INTERRUPT_INTERVAL          0x00000200 ```
Got:
[image]  

### Here is the issue:
For: ``` #define TIMER_INTERRUPT_INTERVAL          0x00000100 ```
Got:
[image]  

With a fast compare int_log i see something did not work as expected:
I ran the tests twice:
- once with disabling the timer interrupts:```    csr_mie = csr_mie | 0x00000808;``` 
- once with enabling the timer interrupts: ```    csr_mie = csr_mie | 0x00000888;``` 

As you can see in this image - only 5 sections are different: (the interrupts) but after the 5th interrupts the test stoped producing more interrupts:
[image]

And there was an instruction that executed on the "interrupts tests" that did not execute in the "with interrupts"
From that point forward, both tests behaved the same (there were no more interrupts taken by the tests that should have taken many more interrupts..
[image]

Instructions to recreate the issue:
1) change this line: ``` #define TIMER_INTERRUPT_INTERVAL          0x00000100 ```
2) Run with interrupts:
``` python build.py -dut core_rrv -cfg core_rrv_boot_trap -top core_rrv_no_ref_tb -tests "bubblesort_lfsr" -app -hw -sim```  
3) change this line: ```    csr_mie = csr_mie | 0x00000808;```  
4) Run without interrupts:
``` python build.py -dut core_rrv -cfg core_rrv_boot_trap -top core_rrv_no_ref_tb -tests "bubblesort_lfsr" -app -hw -sim```  
5) Compare the 2 trk logs:
- target\core_rrv\tests\bubblesort_lfsr\trk_inst.log
- target\core_rrv\tests\bubblesort_lfsr_1\trk_inst.log
  
*Note:  *
I fixed the log generator so it won't override the log after every test. (the logs are generated in the correct directory now)
So all you need to do is:
1) Open both logs
2) ctrl-shift-p
3) compare active file
4) find the "bubblesort_lfsr_1\trk_inst.log" so its compared with "bubblesort_lfsr\trk_inst.log"

amichai-bd, 2024-02-07:
@roman012285 - can we close this issue?
Please review and confirm this is fixed

roman012285, 2024-02-07:
I want to take a look over what we did in details, so lets leave that open for now

amichai-bd, 2024-02-11:
I tested the tests i mentioned:
``` python build.py -dut core_rrv -cfg core_rrv_boot_trap -top core_rrv_no_ref_tb -tests "bubblesort_lfsr" -app -hw -sim```
For: ```#define TIMER_INTERRUPT_INTERVAL          0x00000100```

Got this result:
[image]

I will note that the test is taking some time..
I think we should refactor the test to use the HW version of LFSR instead of SW - I think this would help with the length of the test..

roman012285, 2024-02-12:
I am closing that issue!

closed 2024-02-12 by roman012285