FPGA-MAFIA/fpga_mafia #480: First Issue for Noam & Jacob

amichai-bd, 2024-02-15:
This is your first task!
Please set each bullet once done, when all bullets are marked, close the issue as done
## @NoamRahat :
- [ ] read the README of this repository
- [ ] Have a first read of the wiki TFM
- [ ] see 2 Youtube videos from the README Inspiration 

## @jacobavitan837:
- [ ] read the README of this repository
- [ ] Have a first read of the wiki TFM
- [ ] see 2 Youtube videos from the README Inspiration

amichai-bd, 2024-02-21:
@NoamRahat @jacobavitan837 
I don't see any progress on this task.
The subtask "read the README of this repository" should take less than 5 minutes.

Please update the task when done each subtask

NoamRahat, 2024-02-25:
I have started with the ReadMe but I saw I'm mising some context so I skiped to the TFM welcom

amichai-bd, 2024-02-29:
> I have started with the ReadMe but I saw I'm mising some context so I skiped to the TFM welcome

@NoamRahat  
- Please read the README in full.. 
- Then watch 2 videos.

@jacobavitan837 which 2 videos did you watch?
I recommend the two:
[Tesla: DOJO](https://www.tesla.com/AI) | [Watch the video on Youtube](https://www.youtube.com/watch?v=DSw3IwsgNnc)
[Tenstorrent: Wormhole & more](https://tenstorrent.com/) | [Watch the video on Youtube](https://www.youtube.com/watch?v=32CRYenTcdw)

NoamRahat, 2024-03-06:
I can certainly see the inspiration you've drawn from here!

closed 2024-03-06 by NoamRahat

amichai-bd, 2024-03-06:
> I can certainly see the inspiration you've drawn from here!

@NoamRahat Yup - I hope that now that you have seen the video we will have a common language and a similar understanding of the project's vision!