FPGA-MAFIA/fpga_mafia #482: Install all the TFM of the Mafia project

amichai-bd, 2024-02-15:
Use the wiki to see the list of tools:
https://fpga-mafia.github.io/fpga_mafia_wiki/docs/TFM/welcome

### Very important:
Make sure to document anything that did not work so we can add it to the documentation

amichai-bd, 2024-02-18:
@NoamRahat - from the other issue sounds like you already started working on this.
Please move this issue to the "WIP" Status if you have started working on this

NoamRahat, 2024-02-19:
@amichai-bd 
[image]
for some reason I don't have here the option

amichai-bd, 2024-02-19:
Looks like Roman was able to fix the issue you had.
Let me know if you have any other questions

amichai-bd, 2024-02-21:
@NoamRahat & @jacobavitan837 - Please give an update on what tools you guys were able to install and test and which are still WIP.

NoamRahat, 2024-02-23:
I'm currently at the IDE and extensions stage, making headway. I expect to wrap up by the EOW.

amichai-bd, 2024-02-25:
# @NoamRahat please fill this according to your state
- [ ] | Editor & IDE              | VSCODE                 |  [vs_code](/docs/TFM/projectTool/vs_code.md)        |
- [ ] | Terminal                  | gitbash for windows    |  [GitBash](/docs/TFM/projectTool/GitBash.md)        |            
- [ ] | HW design & verification  | SystemVerilog          |  [SV](/docs/TFM/verilog/intro.md)                   |
- [ ] | HW compiler & simulation  | Modelsim               |  [ModelSIm](/docs/TFM/projectTool/modelsim_intro.md)|
- [ ] | HW FPGA                   | Quartus & DE10-Lite    |  [Quartus](/docs/TFM/projectTool/quartus_intro.md)  |
- [ ] | SW compiler               | RISCV GCC              |  [RISCV GCC](/docs/TFM/projectTool/gcc_intro.md)    |
- [ ] | Scripts & automation      | python                 |  [python](/docs/TFM/projectTool/python.md)          |
- [ ] | Documentation             | Docusaurus             |  [Docusaurus](/docs/TFM/projectTool/docusaurus.md)  |
- [ ] | Diagrams                  | draw.io                |  [Docusaurus](/docs/TFM/projectTool/drawio.md)      |
- [ ] | Version Control           | git                    |  [Git](/docs/TFM/projectTool/intro.md)              |
- [ ] | CI/CD                     | github actions         |  [Git](/docs/TFM/projectTool/intro.md)              |
- [ ] | Project management        | github issues, discussions, projects, milestones, etc |                      |

amichai-bd, 2024-02-25:
# @jacobavitan837 please fill this according to your state
- [ ] | Editor & IDE              | VSCODE                 |  [vs_code](/docs/TFM/projectTool/vs_code.md)        |
- [ ] | Terminal                  | gitbash for windows    |  [GitBash](/docs/TFM/projectTool/GitBash.md)        |            
- [ ] | HW design & verification  | SystemVerilog          |  [SV](/docs/TFM/verilog/intro.md)                   |
- [ ] | HW compiler & simulation  | Modelsim               |  [ModelSIm](/docs/TFM/projectTool/modelsim_intro.md)|
- [ ] | HW FPGA                   | Quartus & DE10-Lite    |  [Quartus](/docs/TFM/projectTool/quartus_intro.md)  |
- [ ] | SW compiler               | RISCV GCC              |  [RISCV GCC](/docs/TFM/projectTool/gcc_intro.md)    |
- [ ] | Scripts & automation      | python                 |  [python](/docs/TFM/projectTool/python.md)          |
- [ ] | Documentation             | Docusaurus             |  [Docusaurus](/docs/TFM/projectTool/docusaurus.md)  |
- [ ] | Diagrams                  | draw.io                |  [Docusaurus](/docs/TFM/projectTool/drawio.md)      |
- [ ] | Version Control           | git                    |  [Git](/docs/TFM/projectTool/intro.md)              |
- [ ] | CI/CD                     | github actions         |  [Git](/docs/TFM/projectTool/intro.md)              |
- [ ] | Project management        | github issues, discussions, projects, milestones, etc |                      |

amichai-bd, 2024-03-03:
@NoamRahat - i think you can mark as done the SystemVerilog, RISCV GCC, python & Project management

@jacobavitan837 Please update on your progress

closed 2024-03-07 by jacobavitan837

amichai-bd, 2024-03-08:
@jacobavitan837 sounds like you got all the TFM ready!
So you should be able to complete #503