FPGA-MAFIA/fpga_mafia #484: .bashrc location

NoamRahat, 2024-02-17:
Hi @amichai-bd ,
I'm now going over the wiki TFM, and I wanted to ask if it matters were the .bashrc file is saved, dose it have to be saved in home dir, or either in the repo's dir.
However I understand I need to create it, becuse I didn't fined it anywhere, correct?

NoamRahat, 2024-02-17:
Out of ChatGPT:
.bashrc file is typically located in your home directory (~/.bashrc). This file is specific to the Bash shell and is executed every time you open a new terminal window or start a new Bash session.
You can create a .bashrc file in your home directory if it doesn't exist already.
While it's conventionally located in the home directory, you can technically place a .bashrc file in other directories if you want to have different configurations for specific projects or contexts. However, it's not common practice to do so, and it's generally more convenient to have a single .bashrc file in your home directory for global settings.

amichai-bd, 2024-02-17:
If I answered your question please mark this as "close with comment"

amichai-bd, 2024-02-17:
@NoamRahat

closed 2024-02-17 by NoamRahat