FPGA-MAFIA/fpga_mafia #491: GCC optimization levels - See why the tests are not passing for -O1, -O3

amichai-bd, 2024-02-19:
Currently in the GCC compile we are running without any optimizations when we are compiling are test.
Will note that we are adding the optimization level -O3 only in the linker and the final compilation, but the Original tests is already in assembly form that I don't know if there are other optimization that can take place when the code is already in assembly form.

See:
```
[COMMAND] riscv-none-embed-gcc.exe -S -ffreestanding -march=rv32i -I ../../../../../app/defines ../../../../.././verif/core_rrv/tests/hw_sw_lfsr_perf.c -o hw_sw_lfsr_perf_rv32i.c.s
[COMMAND] riscv-none-embed-gcc.exe -O3 -march=rv32i -T ../../../../../app/link.common.ld -I ../../../../../app/defines -Wl,--defsym=I_MEM_OFFSET=0 -Wl,--defsym=I_MEM_LENGTH=65536 -Wl,--defsym=D_MEM_OFFSET=65536 -Wl,--defsym=D_MEM_LENGTH=61440 -nostartfiles -D__riscv__ -Wl,-Map=hw_sw_lfsr_perf.map ../../../../../app/crt0/crt0_default.S hw_sw_lfsr_perf_rv32i.c.s -o hw_sw_lfsr_perf_rv32i.elf
```

We need to modify the build to compile the "-S" with optimization.
Test it with -O1, -O2 and -O3.
i think -O1 should be a good starting point
See [descutions ](https://github.com/FPGA-MAFIA/fpga_mafia/discussions/490) To read about it.

amichai-bd, 2024-02-19:
@roman012285  - just FYI

amichai-bd, 2024-02-20:
@roman012285 - FYI
I ran this test before and after updating the build optimization flag:

``` python build.py -dut core_rrv -test alive -app -hw -sim ```

See the example of the elf without any optimization vs -O1 (minimal optimization)
```
[COMMAND] riscv-none-embed-gcc.exe -S -ffreestanding -march=rv32i -I ../../../../../app/defines ../../../../.././verif/core_rrv/tests/alive.c -o alive_rv32i.c.s
```
vs
```
[COMMAND] riscv-none-embed-gcc.exe -O3 -S -ffreestanding -march=rv32i -I ../../../../../app/defines ../../../../.././verif/core_rrv/tests/alive.c -o alive_rv32i.c.s
```

## 2 interesting points:
1) In the "main" -> there is no use of the z so the compiler optimized away all of the logic!
2) in the "sum" -> the compiler found that there is no need for a stack update and simply used the "add" and "ret" 
[image]

amichai-bd, 2024-02-20:
I gave this a try and debug.. and things are confusing. the optimization makes it harder to understand.

First i tried comparing with the reference model.
But then i found out that the reference model is not able to execute the optimized program!
this is the test that i will debug & enable in the reference model. (not the rrv_core).
Once i have that up and running, i will be able to understand whats wrong with the rrv_core by comparing it to the reference model.
``` ./build.py -dut rv32i_ref -test alive_vga -app -hw -sim ```

And this is the test:
[image]

Super simple - but its failing so its great to use for debug

amichai-bd, 2024-02-20:
Found the issue!
The linker has a read-only section that for unknown reason was generated outside the "dmem" region! (0xd14 is outside the DMEM!!)
[image]

We explicitly wrote that the rodata section should be part of the "global_data"
[image]

But for some unknown reason, the -O1 flag moved some of it outside.
ChatGPT suggested fixing the linker SECTIONS a little:
[image]
 
 And now it all works!
 
[image]

amichai-bd, 2024-02-20:
Seems like its failing now the mini_core..

I really tried to understand what happened there..
But this time seems like the compiler just added random 4 reads to regions and location that i definitely did not want him to do..
So i added the cfg option to add the -O1, so currently its enabled in specific cfg..

amichai-bd, 2024-02-20:
I added the optimization only to the "-cfg core_rrv_boot_trap"
See how the run time was cut in half!!!

[image]

vs this:
[image]

amichai-bd, 2024-02-20:
Still need to figure out if we can enable the compile optimization also for the mini_core and big_core..

I don't like it that the compile added 3 reads that are not used.. but i don't know what we can do about it..

roman012285, 2024-02-20:
nice!