FPGA-MAFIA/fpga_mafia #503: HW and SW compilation

roman012285, 2024-02-26:
## Goals 
- You will get familiar with SW/HW compilation and simulation process.
- To make sure that all the required software are installed correctly

## Required tools
- Python
- RISCV GCC
- model sim 
- GitBush

## What we are going to do
Generally speaking, we are going to run one of the `.alive_vga.c` test files on one of our  RISCV cores named `core_rrv`

## Step 1 - cloning the fpga_mafia repository
- In that section we are going to bring the `fpga_mafia` from the git hub to our local computer. 
1. Choose a location where you would like that `fpga_mafia` folders will be.
2. Enter that folder and type
```
git clone  https://github.com/FPGA-MAFIA/fpga_mafia.git
```
3. After Cloning it (bringing from git hub to your computer) type
```
cd fpga_mafia
```
## Step 2 - running sw compilation
- In that section we are going to compile `alive_vga.c` file. This test prints some values as an output.
That test is located in `/verif/core_rrv/test/alive_vga.c`
1. Go to that folder and find that test. 
2. Compile that test by typing the following command
```
./build.py -dut core_rrv -test alive_vga -app
```
make sure you run this command from `fpga_mafia` main folder
3. make sure that the test has passed successfully
[image]

- Pay attention that new folder named `/target` has been created. That folder will contain all the outputs of the building process (sw, hw and simulation) 
4. Go to `/target/core_rrv/tests/alive_vga/gcc_files` and observe their content.
Open the file named `alive_vga_rv32i_elf.txt`. That file is the assembly file of the `alive_vga.c` test

## Step 3 - running hw compilation
- In that section we will compile system verilog files of  the cpu core named `core_rrv`
1. Type the following command
```
./build.py -dut core_rrv -test alive_vga -app -hw
```
make sure the test is passed

## Step 4 - running the simulation
- In that section we will simulate our test `alive_vga`
1. Type the following command
```
 ./build.py -dut core_rrv -test alive_vga -app -hw -sim
```
Observe the created files in the `/target/core_rrv/test/alive_vga` folder. Some of those files contains information about the core functionality and can use us for debug process and tracking after core behavioral.
[image]
2. Open the `screen.log` and try to understand its content.

## Step 4 - running simulation with gui
In that section we will run the gui that shows us the waves of all the signals in the core. Know we will only will check if the gui is openning
1. run the following command
```
 ./build.py -dut core_rrv -test alive_vga -app -hw -sim -gui
```
make sure the gui is opened(That's it for now)

** Let us know if you have any questions **

amichai-bd, 2024-02-28:
@jacobavitan837 @NoamRahat 
Once you you able to run this command:
>  ./build.py -dut core_rrv -test alive_vga -app -hw -sim
Please add a screen-shot as a comment to this issue

NoamRahat, 2024-03-01:
[image]
[image]
[image]
[image]
[image]

amichai-bd, 2024-03-08:
@jacobavitan837 if you have installed all the TFM this task is a single like to execute and finish. Please let me know if you have any questions or issues with this

jacobavitan837, 2024-03-10:
Hi everyone I succeed to run the lines below, I have one warning. the images are arranged in running order 
[image]
[image]
[image]
[image]
[image]

roman012285, 2024-03-10:
nice work!
Put that warning into chat gpt and ask what causes it :-)

jacobavitan837, 2024-03-11:
chat gpt tell me to add more backslash to the path