FPGA-MAFIA/fpga_mafia #504: some questions

jacobavitan837, 2024-02-27:
Hi Amichai , I'm still work on the previous issues but i have some questions.
1. at the second task we need to discuss on the different between asic and fpga you mean to do that as a comment?
2. I have no idea where i need to write the scripts e.g. bash.rc 
3. I'm sorry about the delay

amichai-bd, 2024-02-28:
1. there is a "Discussions" tab in github - press it, add a "general" discussion. Do a 5-10 minutes research asking what ever you want ChatGPT. Then ask ChatGPT to conclude your discussion so you can copy past its answer to the this repo ""Discussions" tab in github".
2. The "bashrc" is not a script. its the bash configuration. (use chatGPT to explain what its used for and what you can do with it..)
    after you installed vscode, gitbash -> add a new file in the path "~/.bashrc" this can be done in many ways in the command 
    line. example ```echo "# Configure bash shell" > ~/.bashrc```
3. let me know if there is anything else i can help with... try and comment directly in the relevant "issues" so the question and answered for that issue will be local

@jacobavitan837

amichai-bd, 2024-02-28:
@jacobavitan837 Once you approve i answered you question, please close this issue

closed 2024-02-28 by jacobavitan837