FPGA-MAFIA/fpga_mafia #532: Add k-t to print function in vga lib

NoamRahat, 2024-03-07:


amichai-bd, 2024-03-07:
[font.zip](https://github.com/FPGA-MAFIA/fpga_mafia/files/14527076/font.zip)
I think file may help -> you should look at this implementation and see how it compares to ours.
Then you can extract the correct method to copy the pixel map to our way of doing this

amichai-bd, 2024-03-07:
https://lpc.opengameart.org/content/8x8-ascii-bitmap-font-with-c-source
[image]

amichai-bd, 2024-03-08:
Added a script to create the bit pixel representation of the ascii char.

Please use it for this task

NoamRahat, 2024-03-09:
under review: https://github.com/FPGA-MAFIA/fpga_mafia/pull/537

amichai-bd, 2024-03-09:
[image]
Seems like this is Done.
I think you can close this issue

NoamRahat, 2024-03-10:
Done.

closed 2024-03-10 by NoamRahat