FPGA-MAFIA/fpga_mafia #546: Test Case for Fibonacci Calculator with Keyboard Input

NoamRahat, 2024-03-11:
Description:

Objective:
The objective of this test case is to verify the functionality of a Fibonacci calculator program that accepts user input from the keyboard and generates Fibonacci numbers accordingly.

# Test Steps:

1. Setup:

- [] Ensure the Fibonacci calculator program is compiled and ready for testing.
- [] Set up the test environment with necessary dependencies.

2. Test Scenario: Keyboard Input for Fibonacci Calculation:

- [] Simulate keyboard input using a test bench.
- [] Inject a number via the keyboard, for example, the number 8.
- [] Ensure the program captures the input correctly using the scanf function.
- [] Verify that the program starts Fibonacci calculation based on the input.

3. Fibonacci Calculation:

- [] Validate that the Fibonacci calculation algorithm is correctly implemented.
- [] Check if the program accurately computes the Fibonacci value for the given input.

4. Output Validation:

- [] Ensure that the calculated Fibonacci value is displayed correctly on the screen.
- [] Verify that the program terminates gracefully after displaying the result.

# Additional Information:

* PS2 Protocol:

- [] Familiarize with the PS2 protocol, which is responsible for communication between the keyboard and the CPU.
- [] Understand the basics of how the keyboard sends signals and how the CPU receives them.

# Test Bench:

- [] Utilize a test bench to simulate keyboard input for the program.
- [] Mimic the behavior of a keyboard by injecting input signals into the program.

# Software Implementation:

- [] Ensure that the program correctly handles keyboard input using the scanf function.
- [] Verify that the program calculates Fibonacci numbers accurately based on the user input.

# Expected Outcome:
- [] Successful execution of the test case should demonstrate that the Fibonacci calculator program accurately calculates Fibonacci numbers based on user input from the keyboard.

amichai-bd, 2024-03-11:
Your plan is well-structured and covers the essential aspects of testing the Fibonacci calculator program. Below are a few clarifications and additional suggestions to enhance your testing strategy:

### Clarifications on Software (SW) Implementation:
- It's great to see you've outlined the process involving the `scanf()` function for keyboard input. Please note, `scanf()` retrieves input as a string, which must then be parsed into an integer. This conversion step is crucial for the Fibonacci calculation to work correctly.
- Aim to encapsulate your logic within functions. This modular approach will facilitate reuse in other projects, as we intend to incorporate this functionality into a library for broader application.

### Test Bench (TB) Development Instructions:
- For simulating keyboard input in your tests, it's necessary to implement both a C file for the test logic and a corresponding ".vh" file to define the keyboard input sequences.
- Specifically, to emulate keyboard input, use the `send_string("7")` command within your ".vh" file. This command simulates the action of typing "7" on the keyboard, which is then processed by your program.
  - Create the test implementation in C: `verif/core_rrv/tests/PS2_alive.c`
  - Define keyboard input in VHDL: `verif/core_rrv/tests/PS2_alive.vh`

- Make sure to add your new tests to the test sequence file at: `verif/core_rrv/tb/hw_seq/core_rrv_hw_seq.vh`

### Handling PS2 Keyboard Inputs:
- Due to current limitations within our testing environment, we cannot fully automate the verification process for tests involving PS2 keyboard inputs. For now, we will rely on visual inspection of the output displayed on the screen to validate test success.

### Running the Test:
- To execute the test without using the reference model (thereby simplifying the verification process), include the `-top core_rrv_no_ref_tb` option in your command. This approach allows for testing in a more controlled environment.

### Summary:
By following these guidelines and structuring your test implementation accordingly, you will enhance the clarity and effectiveness of your testing strategy. This meticulous approach will ensure that the Fibonacci calculator program is rigorously validated, guaranteeing accurate calculations based on user input from the keyboard.

amichai-bd, 2024-03-11:
Watch:
- https://www.youtube.com/watch?v=w1SB9Ry8_Jg&t=340s
- https://www.youtube.com/watch?v=dL0GO9SeBh0

NoamRahat, 2024-03-12:
[image]

NoamRahat, 2024-03-13:
[roman012285](https://github.com/roman012285) added a commit that referenced this issue (https://github.com/FPGA-MAFIA/fpga_mafia/issues/546#ref-commit-ad83ed4)
@roman012285
[Merge pull request](https://github.com/FPGA-MAFIA/fpga_mafia/commit/ad83ed48b0ccac5ca9d15d2506de03c6fc0bb489) https://github.com/FPGA-MAFIA/fpga_mafia/pull/557 [from FPGA-MAFIA/nr_ps2_fibbo](https://github.com/FPGA-MAFIA/fpga_mafia/commit/ad83ed48b0ccac5ca9d15d2506de03c6fc0bb489)