FPGA-MAFIA/fpga_mafia #568: Add PS2_fibbonacci test to rv32i_level0_cfg_intrpt_no_ref file

roman012285, 2024-03-14:
## Introduction
In this issue, you will learn how to integrate your new tests into the sanity check process, which is performed on a remote host computer. This step is crucial to ensure all functionalities are working as expected before any changes are merged into the main branch. Specifically, when you execute `git push origin <your branch>` and create a pull request, it triggers tests that are located in specific directories corresponding to each core, such as `verif/core_rrv/regress` for the core_rrv. However, this process is not limited to just the `core_rrv`; it applies to any core you might be working on, by referencing the appropriate directory where the tests lists for that core are stored.

## Adding the test `PS2_fibbonacci.c`
1. Navigate to `verif/core_rrv/regress/rv32i_level0_cfg_intrpt_no_ref` file. This files contains all the tests that runs with `core_rrv_no_ref_tb.sv` test bench (and more, we will talk about it in the future)
2. Observe the list structure and add you test. Now when you create PR (pull request) your test will be executed for check
THATS IT!!!

### bonus

1.  We have a script that allows you to run the remote tests locally on your computer. This is particularly useful if you're uncertain whether all the tests will pass in the pull request. To perform this local test execution, use the following command:
```
./scripts/sanity_check.py -yml mafia_sanity -dut core_rrv
```
2. Go to `./github/worksflows/mafia_sanity.yml` and try to figure out what's going on there.

NoamRahat, 2024-03-17:
Hi @roman012285 I will suspend this task till 10.4 due to exam period

roman012285, 2024-03-17:
No problem, good luck. Actually its a very short task but of course do whats right for you!

jacobavitan837, 2024-03-20:
Hi roman I add the tests to verif/core_rrv/regress/rv32i_level0_cfg_intrpt_no_ref 
but when i try to run the bonus i receive this message:  
[image]
and in the log file this causing the problem:
[image]

jacobavitan837, 2024-03-20:
Hi Roman I add the line:   #include "interrupt_handler.h"    and the sanity check pass      
[image]

closed 2024-03-20 by jacobavitan837