FPGA-MAFIA/fpga_mafia #621: `core_rrv` has been renamed to `big_core` - ONLY READ ISSUE

roman012285, 2024-04-16:
hi @jacobavitan837  and @NoamRahat. 
The core formerly known as `core_rrv` has been renamed to `big_core`. The organizational structure, including folders, tests, and all files, remains unchanged from its previous incarnation under `core_rrv`.

Please note, the build command referencing `-dut core_rrv` **is no longer valid** . Replace it with big_core in your commands. For example, use the following command to build:
```
./build.py -dut big_core -test alive_vga -hw -sim
```
Please note here that you red this and for the last person reviewing this issue, please ensure to close it.

- [ ]  jacobavitan837
- [ ] NoamRahat

NoamRahat, 2024-04-18:
[image]

roman012285, 2024-04-18:
1.Add the -app flag
2.I think you missing a library that has to he installed. Please work with chatgpt on such errors

roman012285, 2024-04-18:
1.Add the -app flag
2.I think you missing a library that has to he installed. Please work with chatgpt on such errors
3. IMPORTANT - do git pull before doing anything.

NoamRahat, 2024-04-18:
to solve the issue, open a terminal or command prompt and run the following command:

```
pip install termcolor
```

Once the installation is complete, try running your script again, and it should work without any issues.

closed 2024-04-23 by roman012285