FPGA-MAFIA/fpga_mafia #749: Understand mini_core design

roman012285, 2024-11-18:
- [ ] run `example.c` and examine the `main` function  and how `mini_core` performs it
- [ ] Examine the trackers inside the `target/mini_core` directory
- [ ] use the following [line] (https://fpga-mafia.github.io/fpga_mafia_wiki/docs/rvc/intro)
- [ ] add custom instruction `add_inc` for example `add_inc x1, x2, x3 => x1 = x1 + x2 + 1`
use inline assembly `    //asm(".word 0xf4f757a3" : /* outputs / : / inputs / : / clobbers */);`
- [ ] check the functionality of the instruction using the right tackers

abdkeb, 2024-11-20:
Hi Roman,

here is the line of code we added to the exampl.c file:
[image]

Here is the code in binary and the log of the registers:
[image]

closed 2024-11-20 by abdkeb

roman012285, 2024-11-20:
How you prove that it works, I dont see it from the tracker.

abdkeb, 2024-11-20:
here i opened the tracker file in two windows the line is 17c and we can see that prior to the reg x16 was zero after it has 0xf+0xa+1 = 0x1a
[image]

roman012285, 2024-11-20:
Great Job!